gjroots/esp32_nat_router_plus #6: Can't change username/password in web UI

StickyDigit, 2023-07-04:
Have flashed this to a generic Devkit-C. Using Firefox on Linuxmint to configure. I get the same behaviour with Fennec on a smartphone.

When I try to change the admin username and/or password, then hit "SAVE", the screen flashes and the green tick comes up, but the fields are back to admin,123456789.

I seem able to set a static IP and other things just fine. Only the username and password are b0rked.

gjroots, 2023-07-05:
The issue is that changing the admin username and password on the web interface using Firefox or Fennec does not work as expected. Other settings can be configured successfully. To address this, I need more details such as firmware/software version and any troubleshooting steps you've taken. Sharing a GIF or video demonstrating the issue could also help in understanding and resolving it more effectively.

StickyDigit, 2023-07-05:
Okay. Sorry if I was not clear.

I'm going through the motions again and documenting.

First attempt:
OS: Linuxmint 21.1, fully updated yesterday with kernel 6.4.0-060400rc5-generic
Browser: Firefox 114.0.2, private tab.
Other: Generic DevKitC flashed with esp32_nat_router_plus 1.1.2.

1. Log in to "ESP32 Nat Router +" using default username and password. Success.
2. Select "SETTINGS" tab. Sucess.
3. Change text of field Admin Settings, Username, to "nimda".
4. HIt save. Screen flashes, ok tick appears, text has reverted to "admin".
5. Change text of field Admin Settings, Password, to 987654321.
6. HIt save. Screen flashes, ok tick appears, text has reverted to "123456789".
7. Change both fields, then hit save. Both revert.

The same happens on my LineageOS 18.1 androids, browsers: Fulguris v1.8.28, the default LineageOS "Jelly" browser 11, and Fennec 114.2.0.

Troubleshooting/Verification:

I have tried the same process in normal and privacy mode tabs on three machines in four browsers,. I tested in dark and light mode, and tried simple and complex inputs, including changing both fields to their inverse in case upper/lower/special characters would cause issue (nimda 987654321) before submitting. In all cases the data in those fields reverts.

The same is true whether logged in on the AP side, or the STA side.

Changing the username/password to other values, alphanumeric or special characters also fails.

I can set dark mode, turn off the web browser and reactivate with button menu, set AP name and password, literally everything else, all stay changed when save is pressed without issue. The username and password revert to their original values at every attempt.

I tried erasing and re-flashing the ESP32 twice, once configuring the uplink then attempting to change admin user/pass, then again trying to change admin user,pass first. Both those attempts failed too.

I have also tried rebooting with the button bottom right of web-if straight after setting username and password, in case the "visible reversion" was a red-herring, to no avail.

Hopefully I've described this well enough now. It is literally ONLY the admin and 123456789 I cannot successfully change. All other functions of the web interface are doing as expected.

closed 2023-08-27 by gjroots