gjroots/esp32_nat_router_plus #9: Feature Request: MAC Address Filtering

jelle619, 2023-07-08:
**Is your feature request related to a problem? Please describe.**
Your project looks awesome and incredibly interesting. One of the things I would like to use it for, is to connect older devices that don't support modern encryption (e.g. WPA2) to the internet, such as a Nintendo DS. However, currently, to make this work, you would have to create a completely open network, which is not ideal.

**Describe the solution you'd like**
Though of course not a true replacement for WPA2, it would be great if we could use MAC address filtering to regulate which devices may or may not connect to the network. This would allow the network to have some security, rather than none.

**Describe alternatives you've considered**
The only other alternative I could think of is only powering the ESP32 when it is actively in use. Still, when it's in use, anybody could still join the network, which is not ideal.

**Additional context**
https://en.wikipedia.org/wiki/MAC_filtering

gjroots, 2023-07-09:
Hello, @jelle619  Thank you for your interest in the project and the compliment. I want to assure you that I have already planned and designed a filter like this in the past. However, I have not yet released it because I need to find a simple way to save MAC addresses and develop a user-friendly interface to handle all the necessary steps. Currently, I am facing time constraints, which have impacted my focus on this particular aspect. Rest assured, I do plan to release it soon. Thank you for your understanding.

jelle619, 2023-07-11:
Awesome to hear! If I had the experience, I absolutely would have contributed some code myself. However, my experience with micro-controllers such as the ESP32 is unfortunately very limited.

Please take all the time you need. I'm already happy to hear it's being considered. Thanks a ton!

gjroots, 2024-01-26:
Hi @jelle619 
Implemented MAC Filter in the latest update.
please check.....

closed 2024-01-26 by gjroots