gjroots/esp32_nat_router_plus #16: STA password with certain characters needs URL encoding in web i/f.

StickyDigit, 2023-08-27:
Thanks for the 1.1.3 release. Nice work!
On updating, I connected mine to another STA and noticed this...

STA password must have URL encoding if spaces or some other characters involved. Spaces must be replaced with %20, percent sign "%" with %25 etc. I have not tested which characters are problematic other than those cited here.

For example, connecting esp32_nat_router_plus to a STA, the password "This is a 100% good password." must be URL encoded thus "This%20is%20a%20100%25%20good%20password." for entry into the web interface.

**Expected behaviour**
In the above example, it should be possible to enter the password into the web interface without URL encoding.

Not a biggie once you know, and as long as you have reference available.

If not trivial to fix, maybe clues and a table of character->URL-encoded should be added in the web interface for 1.1.4

StickyDigit, 2023-08-27:
The above is true for the "SCAN" tab, which ought to be the quick and easy way to connect to a new AP on the fly.

The issue presents differently in the "SETTINGS" tab, where passwords with spaces work fine without URL percent encoding by just editing the "Advance settings", and changing the SSID and password fields.

BUT, if there's a percent in the password, it must still be URL encoded as %25 to function.

Example (sans quotes).
Entering "This is a 100% cool." into the password field in settings tab will return the page with "This is 100"
Entering "This is 100%25 cool." there will return the page with "This is 100% cool.", and the STA will be connected on reboot.

gjroots, 2023-08-29:
Bug fixed