gjroots/esp32_nat_router_plus #25: ESP32C3 Version

Soforak, 2024-05-17:
**Is your feature request related to a problem? Please describe.**
I tried to change the version to ESP32C3 but couldn't manage to get anything but errors

**Describe the solution you'd like**
Could you please upload a compiled version for the ESP32C3 please?

gjroots, 2024-05-18:
Hi. @Soforak 

Currently i don't have suitable environment setup to build, please share error log may I can help if anything to possible. 
Surely i will share you firmware files as soon as possible.

gjroots, 2024-11-23:
Hi @Soforak, can you please test the firmware and provide me with an update? Thanks!

[esp32C3.zip](https://github.com/user-attachments/files/17879928/esp32C3.zip)

JacobsIntegralEdition, 2025-04-01:
Can u please make the esp32 c3 version for this?

KalleAnka, 2025-06-24:
Hi, now I'm getting two problems:

Warning:
`C:/Ruby34-x64/lib/ruby/gems/3.4.0/gems/bundler-2.4.4/lib/bundler/definition.rb:524:in 'Bundler::Definition#materialize': Could not find google-protobuf-3.24.2-x64-mingw-ucrt in locally installed gems (Bundler::GemNotFound)`
Tried to install it via gem install google-protobuf -v 3.24.2. No problem with install. Restarted VS Code and same problem.

Error:
`Building in release mode
Compiling .pio\build\esp32dev\components\cmd_router\cmd_router.c.o
components/cmd_router/cmd_router.c:15:10: fatal error: spi_flash_mmap.h: No such file or directory
`
Do you know where to get the correct spi_flash_mmap.h file?
Maybe https://github.com/espressif/esp-idf/blob/master/components/spi_flash/include/spi_flash_mmap.h ?

Be advised that I am not yet running the ESP32-C3 compilation, just the standard in platformio.ini.