gjroots/esp32_nat_router_plus #27: ESP32 S3

goanes, 2024-06-18:
Hi
Is it possible to compile a build for ESP32 S3 ?
Thank you.

gjroots, 2024-06-19:
Hi, @goanes. 
I can try, but I currently lack the necessary hardware to test it as these modules aren't available in my country. I'll attempt it, and if successful, I'll share the results with you.

goanes, 2024-06-19:
Hi
Ok thank you. I will test it :)

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

gjroots, 2024-11-23:
Hi @goanes , can you please test the firmware and provide me with an update? Thanks!

[esp32S3.zip](https://github.com/user-attachments/files/17879956/esp32S3.zip)

ycancook, 2025-03-05:
> Hi [@goanes](https://github.com/goanes) , can you please test the firmware and provide me with an update? Thanks!
> 
> [esp32S3.zip](https://github.com/user-attachments/files/17879956/esp32S3.zip)

My esp32-s3-n16r8 device still doesn't work when flashing this firmware

gjroots, 2025-06-24:
Hi, just add these and try to build [env:esp32-s3]

Shere me the result, the rest of code same

Check your device is jtag or usb or else get issue in cmd prompt. That's all

[Esp32s3.zip](https://github.com/user-attachments/files/20874246/Esp32s3.zip)

jimbeam128, 2025-09-17:
@ycancook 

can you give this here a try?

https://github.com/gjroots/esp32_nat_router_plus/issues/39#issuecomment-3301723584

Regards