gjroots/esp32_nat_router_plus #28: ESP32 S3

Ahsan-Himu, 2024-06-28:
**Is your feature request related to a problem? Please describe.**
I've tried to use my ESP32-S3 WROOM-1 (N16R8) as an extender, every time I upload the firmware it seems perfectly uploaded.But when I search for the SSID from any device, it didn't shows up!
After some searching, I realized that, while uploading the firmware, the bluetooth address were added like BLUETOOTH: BHFAAFGH.....,  but the 
WIFI :            address stays empty, I think it is the problem here!

**Describe the solution you'd like**
I would like to use my ESP32 S3 as a Wifi extender (repeater), Deauthar, etc. Please help.

**Additional context**
[image]

gjroots, 2024-06-29:
Hi @Ahsan-Himu 

Please share screen shot of platform.ini file and any other files you modified.

Ahsan-Himu, 2024-07-09:
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
src_dir = main

[env]
platform = espressif32@^6.4.0
framework = espidf
platform_packages =
        framework-espidf @ https://github.com/tasmota/esp-idf/releases/download/v5.1.2-org/esp-idf-v5.1.2-org.zip
board_build.partitions = partitions_example.csv
monitor_filters = direct
monitor_speed = 115200

[env:esp32dev]
board = esp32dev

[env:esp32-c3-devkitm-1]
board = esp32-c3-devkitm-1

[env:esp32-s3]
board = esp32s3box

; [env:esp32-s3-debug]
; extends = env:esp32-s3
; upload_protocol = esp-builtin
; debug_tool = esp-builtin
; build_type = debug

Ahsan-Himu, 2024-07-09:
> Hi @Ahsan-Himu
> 
> Please share screen shot of platform.ini file and any other files you modified.

I have provided full code of the platform.ini

Ahsan-Himu, 2024-07-09:
> Hi @Ahsan-Himu
> 
> Please share screen shot of platform.ini file and any other files you modified.

I have used Espressif's Flash Download Tool , used  esp32_nat_router file by martin ger also.

gjroots, 2024-07-10:
Please give me some time i will look into this issue

gjroots, 2024-07-10:
Sorry for that you provided platform.ini not works for this project, because this project is different from any other, so add this line to file and try again. 

[env:esp32-s3]
board = esp32s3box ; or any other you know
platform = espressif32@5.1.0
framework = espidf

Ahsan-Himu, 2024-07-10:
> Sorry for that you provided platform.ini not works for this project, because this project is different from any other, so add this line to original file and try again.
> 
> [env:esp32-s3] board = esp32s3box ; or any other you know platform = espressif32@5.1.0 framework = espidf

I've tried this but not working, bluetooth and other settings configured but Ethernet address not generated "marked on the SS"

gjroots, 2024-07-10:
You mentioned "Ethernet address" But i don't think this project will work for your board, may be *menu config additional setting required*, even though I don't have test setup for regenerate same issue. So i try to build and share you

gjroots, 2024-11-23:
Hi @Ahsan-Himu , can you please test the firmware and provide me with an update? Thanks!

[esp32S3.zip](https://github.com/user-attachments/files/17879972/esp32S3.zip)

ycancook, 2025-03-05:
> Hi [@Ahsan-Himu](https://github.com/Ahsan-Himu) , can you please test the firmware and provide me with an update? Thanks!
> 
> [esp32S3.zip](https://github.com/user-attachments/files/17879972/esp32S3.zip)

My esp32-s3-n16r8 device still doesn't work when flashing this firmware

JustDaveIII, 2025-07-26:
I have the same problem with an Seeed ESP32-C3, using the ESP Download Tool (flash_download_tool_3.9.9_R2.exe) . I have tried the Full (single file) and the multi-file versions.

-------
Chip Info:
start detect chip...please wait
chip sync ...
Chip is ESP32-C3 (QFN32) (revision v0.3)
Features: Wi-Fi, BT 5 (LE), Single Core, 160MHz, Embedded Flash 4MB (XMC)
Crystal is 40MHz
MAC: 58cf79f3c7c8
Manufacturer: 20
Device: 4016
Status value: 0x600200
Detected flash size: 4MB

---------

Opening a monitor I get:
07:49:43.876 -> invalid header: 0xffffffff
07:49:43.876 -> invalid header: 0xffffffff
07:49:43.876 -> invalid header: 0xffffffff
07:49:43.876 -> invalid header: 0xffffffff
07:49:43.876 -> invalid header: 0xffffffff

-------

JustDaveIII, 2025-07-26:
Uh, I took a look at you page: "How to setup environment and build project"

Nope. Not going to use my time to try to do all that when I have no idea on how to setup / use the five or so different environments. Oh well.