gjroots/esp32_nat_router_plus #30: Request: Support LCD

reukiodo, 2024-11-03:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Would it be possible to utilize an LCD similar to https://www.waveshare.com/esp32-s3-lcd-1.47.htm ?

Some ideas of what to show:
- AP SSID
- password
- IP
- connection status (SSID, signal strength, etc)
- number of attached devices
- maybe even scroll the DHCP hostnames of connected devices

gjroots, 2024-11-05:
Hi @reukiodo. 

It little bit complex, may try but it takes some time.
Will  update it soon....

reukiodo, 2024-11-07:
Would https://www.waveshare.com/product/esp32-s3-geek.htm be any easier? Or just about the same amount of effort?

gjroots, 2024-11-07:
Currently OLED display **ssd1306** is available with me, so I can try.

But your reference item is not available in my country so I will try first available with myself if possible please alter my project based on your need.