gjroots/esp32_nat_router_plus #32: VSCode PlatformIO tools hung on Initializing PlatformIO Core screen

tichaonax, 2024-11-23:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Start from clean repo, clone the repo
2. Open the project with VSCode
3. On the left navigation select PlatformIO tool
4. Initializing PlatformIO Core screen indefinitley

**Expected behavior**
I want to build the binary, expects the list of supported devices from which I can build against.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser [chrome]

**Additional context**
I can open original exp32_nat-Router and other projects successfully

gjroots, 2024-11-23:
It isn't hanging; it's downloading some support software. Please be patient; the speed depends on your internet connection.

tichaonax, 2024-11-23:
It has been like that for two days in a row and then I restarted and it has been like that for 4 hours now

gjroots, 2024-11-23:
Relax, reinstall and clear all data associated with platformio and check again..

tichaonax, 2024-11-23:
Ok thanks I am retrying

tichaonax, 2024-11-23:
It is still running on windows 11 machine so I tried on a MacBookPro and it was successful. However when I build I get the following error

[image]

gjroots, 2024-11-24:
It is due to partition table, i am  preparing but not released

gjroots, 2024-11-24:
Recline and try

tichaonax, 2024-11-24:
Thanks the partition issue resolved the build problem on MAC but on windows 11 still initializing platformIO Core