gjroots/esp32_nat_router_plus #36: Running the router in esp32 long range mode

harueg, 2025-05-11:
Hi
Is there a possibility to run this router in ESP32 lr mode?
I would use it as an range extender.

Thanks

Harald

gjroots, 2025-06-24:
Hi @harueg  i don't think so.. but I tried long time age for test 2nd esp not present so Iam not confirmed it works, and also when you use in lr mode it completely switching so normal wifi not connected on that time it search for lr wifi ssid.

harueg, 2025-06-25:
Hi @gjroots 

Yeah, the challenge is to get the connection on both sides to normal wifi.
I would use wired ethernet. But I've made tests with directional antennas, normal wifi, and can cover a distance of 400m LoS.
Which is enough for the current project.