gjroots/esp32_nat_router_plus #39: Error when Building - The component esp_driver_uart could not be found.

jimbeam128, 2025-08-29:
Hi,

I tried to compile and flash for a ESP32 S3. 

First I tried the binaries to download and flash but no success.

So I tried to compile it and flash afterwards. But when I build, I get component esp_driver_uart could not be found...

I also builded the esp32 nat router from martin-ger (from who you forked). When I compile the source from his code it´s working...
And that code also depends on uart...

Do you have an Idea why it´s "missing"?

gjroots, 2025-08-31:
Hi @jimbeam128 

Please share the log i know the issue but i need to confirm it once.

jimbeam128, 2025-08-31:
hi @gjroots ,

thanks for your reply! Here the logs which were written.

[idf_py_stderr_output_12268.log](https://github.com/user-attachments/files/22065923/idf_py_stderr_output_12268.log)
[idf_py_stdout_output_12268.log](https://github.com/user-attachments/files/22065924/idf_py_stdout_output_12268.log)

gjroots, 2025-08-31:
Clone my repo and replace these files and check once.

[image]

jimbeam128, 2025-09-01:
Hi @gjroots ,

I just use esp-idf (version 5.2.5) alone for trying to compile. No vscode involved.

I even get this error when the target is set to "esp32".

Also when I run idf.py menuconfig, I get the error

`-- Building ESP-IDF components for target esp32
CMake Error at C:/Espressif/Espressif/frameworks/esp-idf-v5.2.5/tools/cmake/build.cmake:288 (message):
  Failed to resolve component 'esp_driver_uart'.`

I mirrored your repo again and replaced the files. I tried to build again, but same error...

When I do it via vscode, I get the same errors. Doesn´t matter if it´s set to esp32 or esp32s3